platformio/platform-lattice_ice40 #1: Update toolchain-icestorm packages

Jesus89, 2016-09-28:
Hi,

In this repo are located the cross-compiling scripts for the [Icestorm toolchain](https://github.com/cliffordwolf/icestorm) used by [Apio](https://github.com/FPGAwars/apio):

https://github.com/FPGAwars/toolchain-icestorm

Please, download the latest release, that contains the packages for:

- Linux_x86_64
- Linux_i386
- Linux_armv7l
- Linux_aarch64
- Windows
- Mac OS

https://github.com/FPGAwars/toolchain-icestorm/releases/tag/0.9

ivankravets, 2016-09-28:
@Jesus89 , @Obijuan thanks a lot for your contribution! 

I contacted a few months ago with @Obijuan and asked to add `package.json` to each toolchain that will contain all information for PlatformIO Package Manager. For example,
```json
{
    "description": "Compiler for FPGA IceStorm",
    "name": "toolchain-icestorm",
    "system": [
        "darwin_x86_64",
        "darwin_i386"
    ],
    "url": "https://github.com/FPGAwars/toolchain-icestorm/releases",
    "version": "1.7.0"
}
```

Could you add this manifest to each archive? In this case, I'll use your binaries. What is more, PlatformIO 3.0 has fully decentrilized architecture. Vendors/Companies/Developers can create own development platform and take control from PlatformIO Core. 

Currently, we use generic package repositories:
- https://github.com/platformio/platform-lattice_ice40/blob/develop/platform.json#L17

Would be good if you create own package storage with own manifest and we switch this development platform to it. Or even, I can give you full access to this repository/or transfer it.

> The first integration step was to add FPGA tools in the PlatformIO project (Platforio-FPGA), which has been always used for classic microcontroller ecosystems. But the slow integration and the lack of flexibility encouraged us to create a custom fork of PlatformIO (pio) called Apio to speed up our investigation and experiments with open FPGAs.

What do you mean under slow integration? Which functionality do you need? Our aim is to develop PlatformIO Core and implement new features. We will see soon "PlatformIO Remote" and other cool features. As practice shows, the forked software doesn't have any perspective. We have big community and I'm sure that it will be interested in FPGA support within PlatformIO.

What is more, PlatformIO is not developed only for "classic microcontroller ecosystems". This is a cross platform build system with platform, package and library managers.  The all functionality could be implemented using own development platform and PlatformIo Build API.

If you need any help with this development platform or have questions, I'm glad to help!

Jesus89, 2016-09-28:
Hi @ivankravets,

Thanks for the information. We will take it into account for the toolchain integration.

Also I will clarify some questions.

> What do you mean under slow integration? Which functionality do you need?

The initial FPGA tools integration started about three months after its final release in 2.9.0. We fully understand that there were a lot of work and more major issues, but during this time we could research, develop and distribute Apio as a tool compatible with these new FPGA toolchain. And that's why this project started.

Also, until 3.0.0 release in September, there weren't flexible features like decentralized architecture. Now I think we can take advantage of this functionality for the contributions and new platforms support.

> Our aim is to develop PlatformIO Core and implement new features. We will see soon "PlatformIO Remote" and other cool features. As practice shows, the forked software doesn't have any perspective. We have big community and I'm sure that it will be interested in FPGA support within PlatformIO.

We are agree in spreading FPGAs into the community, that's why we started FPGAs support in PlatformIO. However, Apio is not just a fork. It is a tool for doing experiments, and for including whatever we need about open FPGA experimentation. It shares some features with PlatformIO, but it contains other different features like external tools (lsusb, lsftdi), drivers configuration, boards detection and more things that make our research easier.

I know that PlatformIO is developed for generic platforms, beyond microcontrollers, but new concept platforms, like FPGA, require new definitions, new problem solving and experimentation. Even toolchain-icestorm package is not a fully stable toolchain because some of its sub-projects don't have official releases and they are in a continuous development.

I'm sure that every step will be beneficial to the open source community.

Thanks for your help and your work!

ivankravets, 2016-09-28:
How is difficult for your team to support this repo? is it possible? How do you see that?

ivankravets, 2016-09-28:
> but it contains other different features like external tools (lsusb, lsftdi), drivers configuration, boards detection and more things that make our research easier.

You decide which tools development platform will use. Also, you decide where these tools will be stored.

If you need to perform extra steps from user's side, you can implement extra `target` in own dev.platform and attached specific actions. For example, `apio drivers` (install some drivers), it is equal to `pio run -t drivers`. Then in `dev.platform/scripts/main.py` (bottom) just add new target `AlwaysBuild(env.Alias("drivers", "/path/to/drivers/source", actions))`.

Actions could be different: SCons command, Python function and etc. These things are documented in SCons docs.

I'm ready to help, but I don't know what to do in this case. I don't have practice with FPGA.

Jesus89, 2016-09-28:
> How is difficult for your team to support this repo? is it possible? How do you see that?

We can contribute with new features and improvements, and our boards support. We can generate new packages releases when required. However, this is not our main focus, we are also doing teaching materials, tutorials, workshops, new FPGA boards, a graphic FPGA editor and more.

For other FPGA boards support, OS bugfixing, etc, we rely on the community.

> I'm ready to help, but I don't know what to do in this case. I don't have practice with FPGA.

Thanks for your help! @Obijuan and I are going to discuss and organize the way to contribute, and also to encourage the new community of open FPGAs to contribute. We are at an early stage with unstable new tools and a fast growing. We must be tuned to learn and integrate these tools.

Regards.

closed 2016-10-02 by Jesus89