platformio/platform-lattice_ice40 #2: Add toolchain-iverilog packages

Jesus89, 2016-09-28:
Hi,

In this repo are located the cross-compiling scripts for the [Icarus Verilog toolchain](http://iverilog.icarus.com/) used by [Apio](https://github.com/FPGAwars/apio):

https://github.com/FPGAwars/toolchain-iverilog

Please, download the latest release, that contains the packages for:

- Linux_x86_64
- Linux_i686
- Linux_armv7l
- Linux_aarch64
- Windows
- Mac OS

https://github.com/FPGAwars/toolchain-iverilog/releases/tag/v0.3

Jesus89, 2016-10-02:
Hi,

I have created this file https://github.com/FPGAwars/toolchain-iverilog/blob/master/manifest.json, and added its reference to this repo's platform.json.

I obtain Error: Detected unknown package.

Is there some semantic/strong relation between tag, version and package name?

I have used sha1sum over the package file. Is this correct?

Thanks.

ivankravets, 2016-10-02:
It should be `package.json`. Check any packages for PlatformIO for example.

ivankravets, 2016-10-02:
1. Currently, package manager supports `zip` and `tar.gz` formats (Python-native, for cross-platform).
2. Package should contain `package.json` manifest. Please check any PIO3 package.

Jesus89, 2016-10-03:
Thanks. I will rebuild the packages with this format.

Jesus89, 2016-10-03:
I have just rebuilt all the packages with the required format: https://github.com/FPGAwars/toolchain-iverilog/releases/tag/v1.0.0.

A last question. For future versions it is better to add those new packages to the manifest, or to substitute the version fields. I have observed both behaviors.

ivankravets, 2016-10-03:
Thanks!

> For future versions it is better to add those new packages to the manifest, or to substitute the version fields.

It depends on a situation. If released development platform depends on version `1.0.0` and you are working on `dev` version of development platform that depends on `1.0.1`, in this case, you need to have 2 versions in manifest.

There is no problem when many versions will be defined in a manifest. PlatformIO Package Manager will look for the best version which matches with requirements in `platform.json`.

ivankravets, 2016-10-03:
I think that we can remove all base package manifests from [here](https://github.com/platformio/platform-lattice_ice40/blob/develop/platform.json#L17) and define your manifests.

Jesus89, 2016-10-03:
Ok.

> I think that we can remove all base package manifests from here and define your manifests.

Yes, thats what I am testing in my fork: https://github.com/Jesus89/platform-lattice_ice40/blob/iverilog/platform.json#L21. I prefer by now to keep the old manifests because of the toolchain-icestorm packate, that will be updated with 1.10.0 (https://github.com/FPGAwars/toolchain-icestorm/issues/63) because it will be a stable release.

ivankravets, 2016-10-03:
https://github.com/Jesus89/platform-lattice_ice40/commit/a7aa0fa45d771a3f1e9eabc5234d528eb67c01d3#diff-7bf8a7e6a6e69b354a3d3e106cc0cbacR39

Do you have a problem on Windows? OS should resolve it automatically.

Jesus89, 2016-10-03:
I check Windows to resolve .exe extensions. Also it is required for icarus verilog and vvp execution arguments.

Jesus89, 2016-10-03:
I will update it to use Windows check only for icarus verilog execution (https://github.com/platformio/platform-lattice_ice40/issues/4)

ivankravets, 2016-10-03:
I asked because we have tons of Windows binaries and don't use this checking.

I think that SCons handles this calling when processing Action.

Jesus89, 2016-10-03:
Yes, thanks for the tip!

closed 2016-10-03 by Jesus89