platformio/platform-lattice_ice40 #4: Icarus Verilog support

Jesus89, 2016-09-28:
Using [toolchain-verilog](https://github.com/FPGAwars/toolchain-iverilog) package requires passing some paths to the binaries **iverilog** and **vvp** because it is a local installation:

- _IVL_: /path/to/toolchain-iverilog/lib/ivl
- _VLIB_: /path/to/toolchain-iverilog/vlib/system.v  (yosys lib compatibility)

Use cases:

- Linux/Darwin:
 - iverilog -B _IVL_ _VLIB_ sources
 - vvp -M _IVL_ source
- Windows:
 - iverilog.exe _VLIB_ sources
 - vvp.exe source

There is a standard way to pass these paths to the builder?

ivankravets, 2016-09-28:
Yes, it's possible. With PlatformIO 3.0 you have full control on PlatformIO. In this case, in main script `main.py` need to add
```python
from os.path import join

platform = env.PioPlatform()
env.PrependENVPath("PATH", join(platform.get_package_dir("toolchain-verilog ", "lib", "ivl")))
```

In this case, PlatformIO Core will manage packages using semantic versioning and depended packages will be automatically loaded.

Jesus89, 2016-09-28:
Ok! Thanks for the information.

ivankravets, 2016-10-03:
I granted you with full permissions to this repo. You can work on it without PR.

Jesus89, 2016-10-03:
Great. Thanks!

Jesus89, 2016-10-03:
I suppose that correct syntax is:

`join(platform.get_package_dir("toolchain-verilog"), "lib", "ivl"))`

instead of `join(platform.get_package_dir("toolchain-verilog ", "lib", "ivl"))`

Also, I can not access to *platform.get_package_dir("toolchain-iverilog")* because there is no toolchain-iverilog version in appstate.json although package is installed. Same with toolchain-icestorm.

ivankravets, 2016-10-03:
> join(platform.get_package_dir("toolchain-verilog"), "lib", "ivl"))

This is correct syntax.

>  platform.get_package_dir("toolchain-iverilog") 

While package isn't installed yet, need to use `join(platform.get_package_dir("toolchain-verilog") or "", "lib", "ivl"))`

> appstate.json

PlatformIO 3.0 doesn't use `appstate.json` for package registry. It relays on real manifests located in the packages. Please check that `toolchain-iverilog` has `package.json`.

Jesus89, 2016-10-03:
Yes. It contains the `package.json`. Just everything reinstalled and it works.

closed 2016-10-03 by Jesus89