platformio/platform-lattice_ice40 #5: Ready for 1.1.0

Jesus89, 2016-10-03:
Hi,

I have created the release/v1.1.0:
* Targets: verify, sim, time
* Improved examples

Tests passed. Can I merge in master?

ivankravets, 2016-10-03:
I recommend using [git-flow](http://nvie.com/posts/a-successful-git-branching-model/) that manages releases/features automatically.

1. Remove manually created release branch
2. Switch to `develop` branch
3. Start new release
4. Increase platform version in `platform.json` manifest
5. Finish release

`git-flow` will do rest work automatically.

ivankravets, 2016-10-03:
Also, I can make releases if you don't want to paly with this git-flow.

Jesus89, 2016-10-03:
Great. git-flow is a nice way.

I can manage releases. Thanks!

closed 2016-10-03 by Jesus89

ivankravets, 2016-10-03:
Do you need any help?

Jesus89, 2016-10-03:
I have created the release, but develop branch has been merged from master instead of release/v1.1.0. Is there a git-flow configuration to enable merging develop from release branch?

ivankravets, 2016-10-03:
Have you done these steps? https://github.com/platformio/platform-lattice_ice40/issues/5#issuecomment-251194473

You should start all work from `develop`, not `master`. You should not touch `master`. `git-flow` will automatically merge changes to master and rebase `develop` to the latest commit.

ivankravets, 2016-10-03:
Nice! It looks that you forgot to push `tag` to remote.

Jesus89, 2016-10-03:
Yes. Tag pushed: https://github.com/platformio/platform-lattice_ice40/releases/tag/v1.1.0.

Jesus89, 2016-10-03:
I can not understand why develop is merged from master instead of the release branch.

```
$ git flow release start 'v1.1.0'
Switched to a new branch 'release/v1.1.0'

Summary of actions:
- A new branch 'release/v1.1.0' was created, based on 'develop'
- You are now on branch 'release/v1.1.0'

Follow-up actions:
- Bump the version number now!
- Start committing last-minute fixes in preparing your release
- When done, run:

     git flow release finish 'v1.1.0'
```

```
$ git flow release finish 'v1.1.0'
Switched to branch 'master'
Your branch is up-to-date with 'origin/master'.
Merge made by the 'recursive' strategy.
...
Switched to branch 'develop'
Your branch is up-to-date with 'origin/develop'.
Merge made by the 'recursive' strategy.
...
Deleted branch release/v1.1.0 (was fdb7e14).

Summary of actions:
- Release branch 'release/v1.1.0' has been merged into 'master'
- The release was tagged 'v1.1.0'
- Release tag 'v1.1.0' has been back-merged into 'develop'
- Release branch 'release/v1.1.0' has been locally deleted
- You are now on branch 'develop'

```

ivankravets, 2016-10-03:
Please read http://nvie.com/posts/a-successful-git-branching-model/

`master` branch is used ONLY for RELEASES. We should not code here. The latest branch is `develop`.

Jesus89, 2016-10-03:
Yes. I know that. I mean about git-flow automatic tasks. I will investigate that.

Jesus89, 2016-10-03:
The issue I have observed depends on the git-flow version. My git-flow (apt installed) version has different behavior than yours. Simply that.

Jesus89, 2016-10-03:
Fixed using:

`$ git config --global gitflow.release.finish.nobackmerge yes`

https://github.com/petervanderdoes/gitflow-avh/blob/develop/git-flow-release#L136