platformio/platform-lattice_ice40 #6: Cannot override uploader command

kitling, 2017-01-31:
There are various different methods of 'uploading' a bitstream, they depend on your setup.

For instance, with the mystorm board, which has a hardcore cpu as well, the bitstream is managed by the hardcore, with methods ranging from 'include this in the hardcore binary' to 'just send it out over serial and the hardcore will handle it'. I could also imagine a 'stick this on an SD card' upload or others.

At the moment, there is no way to override the uploader command from an extra_script, so if one *really* wants to change the behavior, they have to prepend PATH with a directory that contains a fake 'iceprog' script that actually does what they want.

Alternatively, there may be some SCons magic that I'm not familiar with and don't know how to find. (I'm not a SCons user, but I've been looking into it due to platformio.)

Jesus89, 2017-02-01:
A solution may be to allow override the programmer in an environment variable (Just editing the builder).
Let's see what Ivan thinks.

ivankravets, 2017-02-02:
@Jesus89 Could you implement uploading using a separate command? Please take a look how we do that for Atmel AVR development platform:
* https://github.com/platformio/platform-atmelavr/blob/develop/builder/main.py#L182

Here is example how user can override this "upload command"
* http://docs.platformio.org/en/latest/projectconf.html#custom-uploader

Jesus89, 2017-03-01:
Added in the builder the env vars:
- $UPLOADER
- $UPLOADERFLAGS
- $UPLOADBINCMD = $UPLOADER $UPLOADERFLAGS $SOURCES

These variables can be modified from an extra_script.py:

```python
Import('env')
env.Replace(UPLOADBINCMD='yourprogrammer $SOURCES')
```

```python
Import('env')
env.Replace(UPLOADER='yourprogrammer', UPLOADFLAGS='yourflags')
```

It will be available in the release v1.2.0

closed 2017-03-01 by Jesus89

ivankravets, 2017-03-01:
@Jesus89 Thanks a lot for your contribution! Can I release 1.2.0 or you have other changes?

Jesus89, 2017-03-01:
I am doing more changes. Btw, I have included windows_x86 and windows_amd64 packages in each toolchain (icestorm, iverilog) but AppVeyor looks for windows package. Do I need to fix the manifest.json?

Jesus89, 2017-03-01:
Thanks @ivankravets. https://github.com/FPGAwars/toolchain-icestorm/pull/64