platformio/platform-lattice_ice40 #7: dyld: Library not loaded: /usr/local/opt/libffi/lib/libffi.6.dylib

ivankravets, 2018-02-05:
```
yosys -p "synth_ice40 -blif .pioenvs/icestick/hardware.blif" -q src/counter.v
dyld: Library not loaded: /usr/local/opt/libffi/lib/libffi.6.dylib
Referenced from: /Users/ikravets/.platformio/packages/toolchain-icestorm/bin/yosys
Reason: image not found
```

ivankravets, 2018-02-05:
@Jesus89 do you have an updated toolchain?

Jesus89, 2018-02-06:
Hi @ivankravets!

The toolchain is the same. You need to install `libffi` in Mac OS in order to execute yosys. The same for `libftdi` to enable uploading the bitstream to the FPGA.

I use `brew` to install these packages.

ivankravets, 2018-02-06:
Do you need header files of `libffi`? If no, we can put `libffi.6.dylib` directly to that package in the folder where binary file is located. In this case, no need to install `brew` or `libffi`. What do you think?

jcw, 2018-03-13:
Yep, can confirm, same issue here. Solved with `brew install libffi`. No headers involved, AFAICT.

abuisman, 2018-09-14:
In my case linking libffi (which was already installed, but unlinked) fixed the issue

ivankravets, 2018-11-03:
Resolved in v1.2.2. Please `pio update`.

- https://github.com/platformio/platform-lattice_ice40/releases/tag/v1.2.2

ivankravets, 2018-11-06:
@jcw @abuisman could run `brew uninstall libffi` and run `pio update`? Does it work now?

hqms, 2018-11-15:
> In my case linking libffi (which was already installed, but unlinked) fixed the issue

or you can export variables to determine the libraries

libffi is keg-only, which means it was not symlinked into /usr/local,
because some formulae require a newer version of libffi.

For compilers to find libffi you may need to set:
  export LDFLAGS="-L/usr/local/opt/libffi/lib"

For pkg-config to find libffi you may need to set:
  export PKG_CONFIG_PATH="/usr/local/opt/libffi/lib/pkgconfig"

ivankravets, 2018-11-15:
@hqms what is your macOS version? Do you use the latest version of dev/platform?

sqaview, 2019-03-15:
I'm running macOS Mojave, 10.14.2
I had to uninstall libffi, then reinstall it:
```
brew uninstall libffi
brew install libffi
```

And then, set the env vars in my .bash_profile as indicated above:
```
# homebrew required vars
export LDFLAGS="-L/usr/local/opt/libffi/lib"
export PKG_CONFIG_PATH="/usr/local/opt/libffi/lib/pkgconfig"
```

That worked 👍

tomachinz, 2019-07-28:
I got it working after a re-install of libffi, without changing my env vars. 

I got this message about Brew refusing to link macOS-provided software on the way but all was well... 
```

➜   brew install libffi
Updating Homebrew...
Warning: libffi 3.2.1 is already installed, it's just not linked
You can use `brew link libffi` to link this version.
➜   brew link libffi
Warning: Refusing to link macOS-provided software: libffi
For compilers to find libffi you may need to set:
  export LDFLAGS="-L/usr/local/opt/libffi/lib"

For pkg-config to find libffi you may need to set:
  export PKG_CONFIG_PATH="/usr/local/opt/libffi/lib/pkgconfig"
➜   brew uninstall libffi
➜   brew install libffi
Uninstalling /usr/local/Cellar/libffi/3.2.1... (16 files, 296.9KB)
==> Downloading https://homebrew.bintray.com/bottles/libffi-3.2.1.mojave.bottle.tar.gz
######################################################################## 100.0%
==> Pouring libffi-3.2.1.mojave.bottle.tar.gz
==> Caveats
libffi is keg-only, which means it was not symlinked into /usr/local,
because some formulae require a newer version of libffi.

For compilers to find libffi you may need to set:
  export LDFLAGS="-L/usr/local/opt/libffi/lib"

For pkg-config to find libffi you may need to set:
  export PKG_CONFIG_PATH="/usr/local/opt/libffi/lib/pkgconfig"

==> Summary
🍺  /usr/local/Cellar/libffi/3.2.1: 16 files, 296.8KB

```
Seems to work! 

```
➜   ffmpeg
ffmpeg version 4.1.4 Copyright (c) 2000-2019 the FFmpeg developers
  built with Apple LLVM version 10.0.1 (clang-1001.0.46.4)
  configuration: --prefix=/usr/local/Cellar/ffmpeg/4.1.4_1 --enable-shared --enable-pthreads --enable-version3 --enable-avresample --cc=clang --host-cflags='-I/Library/Java/JavaVirtualMachines/adoptopenjdk-12.0.1.jdk/Contents/Home/include -I/Library/Java/JavaVirtualMachines/adoptopenjdk-12.0.1.jdk/Contents/Home/include/darwin' --host-ldflags= --enable-ffplay --enable-gnutls --enable-gpl --enable-libaom --enable-libbluray --enable-libmp3lame --enable-libopus --enable-librubberband --enable-libsnappy --enable-libtesseract --enable-libtheora --enable-libvorbis --enable-libvpx --enable-libx264 --enable-libx265 --enable-libxvid --enable-lzma --enable-libfontconfig --enable-libfreetype --enable-frei0r --enable-libass --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-librtmp --enable-libspeex --enable-videotoolbox --disable-libjack --disable-indev=jack --enable-libaom --enable-libsoxr
  libavutil      56. 22.100 / 56. 22.100
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100
  libpostproc    55.  3.100 / 55.  3.100
Hyper fast Audio and Video encoder
usage: ffmpeg [options] [[infile options] -i infile]... {[outfile options] outfile}...

Use -h to get full help or, even better, run 'man ffmpeg'
```

manjeethike, 2019-10-29:
> I'm running macOS Mojave, 10.14.2
> I had to uninstall libffi, then reinstall it:
>
> ```
> brew uninstall libffi
> brew install libffi
> ```
>
> And then, set the env vars in my .bash_profile as indicated [...]

neowinston, 2020-04-22:
In my case I had installed both **libffi 3.2.1** and **libffi 3.3**, then I switched from 3.3 back to 3.2.1 like so:

`brew switch libffi 3.2.1`

pedrorrivero, 2020-04-26:
As of today, `dyld: Library not loaded: /usr/local/opt/libffi/lib/libffi.6.dylib` pops up when calling **ffmpeg** on a fresh install of **libffi**. 

`/usr/local/opt/libffi/lib/libffi.6.dylib` does not exist anymore apparrently, it has been renamed to `/usr/local/opt/libffi/lib/libffi.7.dylib`.

I fixed the issue by running:
```
cp 
/usr/local/opt/libffi/lib/libffi.7.dylib /usr/local/opt/libffi/lib/libffi.6.dylib
```

Notice however that  when **libffi** is installed through hombrew (i.e. `brew install libffi`), it creates a symlink `/usr/local/opt/libffi/lib/libffi.dylib` to `/usr/local/opt/libffi/lib/libffi.7.dylib`. That suggests that **ffmpeg** should start using this alternative path instead to avoid any errors.

saleh-mir, 2020-04-26:
I ended up fixing it by downgrading libffi using this guide:
https://dae.me/blog/2516/downgrade-any-homebrew-package-easily

freeslugs, 2020-04-29:
yeah man it's just a different version libffi.6.dylib vs libffi.7.dylib

freeslugs, 2020-04-29:
This will do the trick
```
cd /usr/local/opt/libffi/lib
ln -s libffi.7.dylib libffi.6.dylib
```
then try `ffmpeg`

diegoulloao, 2020-05-29:
> This will do the trick
> 
> ```
> cd /usr/local/opt/libffi/lib
> ln -s libffi.7.dylib libffi.6.dylib
> ```
> 
> then try `ffmpeg`

Yes, that works!!!! thanks.

tupizz, 2020-06-04:
> In my case I had installed both **libffi 3.2.1** and **libffi 3.3**, then I switched from 3.3 back to 3.2.1 like so:
> 
> `brew switch libffi 3.2.1`

tyyy

shengchaoqin, 2020-06-08:
> As of today, `dyld: Library not loaded: /usr/local/opt/libffi/lib/libffi.6.dylib` pops up when calling **ffmpeg** on a fresh install of **libffi**.
>
> `/usr/local/opt/libffi/lib/libffi.6.dylib` does [...]

Thank you very much indeed! Your suggestion
 "cp  /usr/local/opt/libffi/lib/libffi.7.dylib /usr/local/opt/libffi/lib/libffi.6.dylib" solved my problem (after spending two nights trying to install why3 (before frama-c) but got stuck due to this problem. Does anybody know why not everyone has this issue? Why ....6.dylib is required by there is only a ....7.dylib? Three of us installed the same thing and the other two didn't have this issue at all

gilles6, 2020-06-30:
I fixed it this way on macos:
```
brew reinstall php
```

TianyiShi2001, 2020-07-11:
> This will do the trick
> 
> ```
> cd /usr/local/opt/libffi/lib
> ln -s libffi.7.dylib libffi.6.dylib
> ```
> 
> then try `ffmpeg`

That magically works! Thank you!

obrien-k, 2020-08-14:
`brew reinstall libffi` fixed this for me.

closed 2020-08-20 by Jesus89

sosaucily, 2021-01-09:
For me this error was being thrown when running pod install on a Flutter project. 

I needed to gem install a newer version of ffi.

kumardurgesh91, 2021-03-22:
Updating `libffi`worked for me on macOS catalina 10.15.6
`brew install libffi
Error: libffi 3.3_2 is already installed.
To upgrade to 3.3, run:
  brew upgrade libffi
❯
❯ brew upgrade libffi
==> Upgrading 1 outdated package:
libffi 3.3_2 -> 3.3_3
==> Upgrading libffi 3.3_2 -> 3.3_3
==> Downloading https://homebrew.bintray.com/bottles/libffi-3.3_3.catalina.bottle.tar.gz
######################################################################## 100.0%
==> Pouring libffi-3.3_3.catalina.bottle.tar.gz
==> Caveats
libffi is keg-only, which means it was not symlinked into /usr/local,
because macOS already provides this software and installing another version in
parallel can cause all kinds of trouble.

For compilers to find libffi you may need to set:
  export LDFLAGS="-L/usr/local/opt/libffi/lib"
  export CPPFLAGS="-I/usr/local/opt/libffi/include"

For pkg-config to find libffi you may need to set:
  export PKG_CONFIG_PATH="/usr/local/opt/libffi/lib/pkgconfig"

==> Summary
🍺  /usr/local/Cellar/libffi/3.3_3: 17 files, 540.3KB
Removing: /usr/local/Cellar/libffi/3.2.1... (16 files, 297.0KB)
Removing: /usr/local/Cellar/libffi/3.3_2... (17 files, 540.5KB)
==> Upgrading 7 dependents:
cairo 1.16.0_2 -> 1.16.0_5, glib 2.62.0 -> 2.68.0, gnutls 3.6.9 -> 3.6.15, harfbuzz 2.6.1 -> 2.8.0, libass 0.14.0_1 -> 0.15.0, p11-kit 0.23.17 -> 0.23.22, ffmpeg 4.1.4_2 -> 4.3.2_1
==> Upgrading cairo 1.16.0_2 -> 1.16.0_5
==> Downloading https://homebrew.bintray.com/bottles/gdbm-1.19.catalina.bottle.tar.gz
==> Downloading from https://d29vzk4ow07wi7.cloudfront.net/a3e43170a1d8413e6817e57b7218828af22a20b2221d804ad529a68248840a51?response-content-dispositi
######################################################################## 100.0%
==> Downloading https://homebrew.bintray.com/bottles/mpdecimal-2.5.1.catalina.bottle.tar.gz
==> Downloading from https://d29vzk4ow07wi7.cloudfront.net/1a8314428019cec85756be0ea10bc4703cd754ef78a4cb560ddcc559af616a72?response-content-dispositi
######################################################################## 100.0%
==> Downloading https://homebrew.bintray.com/bottles/sqlite-3.34.1.catalina.bottle.tar.gz
==> Downloading from https://d29vzk4ow07wi7.cloudfront.net/63b13860a58b273821c8af6bcd09ed1d38aa9c95f51192257b29f7512fb38859?response-content-dispositi
######################################################################## 100.0%
Error: python@3.9: the bottle needs the Apple Command Line Tools to be installed.
  You can install them, if desired, with:
    xcode-select --install
..........
.........
dd the following to your .bashrc or equivalent:
  export GUILE_LOAD_PATH="/usr/local/share/guile/site/3.0"
  export GUILE_LOAD_COMPILED_PATH="/usr/local/lib/guile/3.0/site-ccache"
  export GUILE_SYSTEM_EXTENSIONS_PATH="/usr/local/lib/guile/3.0/extensions"
==> unbound
To have launchd start unbound now and restart at startup:
  sudo brew services start unbound
==> gnutls
If you are going to use the Guile bindings you will need to add the following
to your .bashrc or equivalent in order for Guile to find the TLS certificates
database:
  export GUILE_TLS_CERTIFICATE_DIRECTORY=/usr/local/etc/gnutls/
`

viertelb, 2021-12-13:
This helped me use my old `easy-tag` on High Sierra after brew updated.

agilebean, 2021-12-29:
For M1 (Apple Silicon) Chips, the libraries are stored on different paths.
So linking is done via:

```
# libffi linking
export LDFLAGS="-L/opt/homebrew/opt/libffi/lib"
export PKG_CONFIG_PATH="/opt/homebrew/opt/libffi/lib/pkgconfig"
```

> As of today, `dyld: Library not loaded: /usr/local/opt/libffi/lib/libffi.6.dylib` pops up when calling **ffmpeg** on a fresh install of **libffi**.
>
> `/usr/local/opt/libffi/lib/libffi.6.dylib` does [...]

dragon-yuan, 2022-01-21:
macOS 12.1 Monterey

cd /usr/local/opt/libffi/lib
ln -s libffi.8.dylib libffi.6.dylib

asad-shakil, 2022-09-30:
> same issue but no this lib, I am using rails in mac m1.
> 
> already done `brew install libffi`
> 
> `rails --help` got a similar error.
> 
> ```
> Library not loaded: /opt/homebrew/opt/libffi/lib/libffi.7.dylib (LoadError)
> ```
> 
> fixed with this:
> 
> ```
> cp /opt/homebrew/opt/libffi/lib/libffi.8.dylib /opt/homebrew/opt/libffi/lib/libffi.7.dylib
> ```
> 
> thanks the above answer

I've tried every other solutions but only this one worked for My M1 Mac.

Dorapopy, 2023-06-12:
fixed this for me:
pip3 install --force-reinstall cryptography

edmondja, 2024-02-29:
> fixed this for me: pip3 install --force-reinstall cryptography

I love you dude, nothing works besides your method.

Dorapopy, 2024-02-29:
裴雪琴已收到，会尽快回复

commit f72e881c (2018-11-03): Fix an issue with "dyld: Library not loaded: libffi.6.dylib" // Resolve #7